oxan/esphome-stream-server #1: Support for UART hardware flow control (serial data flow control)?

Hedda, 2021-05-20:
Does this stream server custom component have support for UART hardware flow control (serial data flow control)?

This feature has also been requested here -> https://github.com/esphome/feature-requests/issues/1126

PS: Example of MCU serial devices that can support hardware flow control with the correct firmware is Zigbee SMD modules. Ex:

https://github.com/tube0013/tube_gateways

https://www.tubeszb.com/shop/coordinators/2

https://github.com/zigpy/zigpy/discussions/584

https://github.com/zigpy/zigpy/discussions/743

oxan, 2021-05-20:
Not specifically. This component doesn't do anything UART-specific at all, it leaves that to ESPHome. So theoretically if ESPHome sets up the UART with hardware flow control, it should work.

oxan, 2021-05-30:
I'm going to close this, as I don't think there's anything actionable here.

closed 2021-05-30 by oxan