oxan/esphome-stream-server #3: ESP32 UART2 Configuration

cybertza, 2021-08-05:
Hi 

Thanks for taking the time to create a package for this.

I am moving from an Serial2Wifi Bridge as per 
https://github.com/ParadoxAlarmInterface/pai/wiki/Connection-methods#serial-over-ip-esp32

I originally just used the  https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino

But since i enjoy breaking thigs I'm trying to move onto ESPhome for this.

I cant really see why this wont work.

If i connect to the streaming port thou, i get some interesting data, from recollection, if i used to telnet to the other port, i used to get text data.
```
external_components:
  - source: github://oxan/esphome-stream-server
uart:
  id: uart_bus
  # add further configuration for the UART here
  baud_rate: 9600
  tx_pin: 17
  rx_pin: 16
  # tx2 and rx2
stream_server:
  uart_id: uart_bus
  # port: 1234
```

So my quesitons are as follows:
1) can this do what i need?
2) Is there a way to directly specify uart2 of is specifying the pins ok?

cybertza, 2021-08-05:
So i guess this basically falls under documentation examples :p

oxan, 2021-08-05:
Your configuration looks correct to me. Are you sure the baudrate is correct?

plambrechtsen, 2021-09-29:
I have just successfully used this configuration on my Wemos D1 ESP32 with UART2. And I had to make sure the TX/RX was around the right way.

```
external_components:
  - source: github://oxan/esphome-stream-server

uart:
   id: uart_bus
   tx_pin: GPIO17
   rx_pin: GPIO16
   baud_rate: 38400

stream_server:
   uart_id: uart_bus
   port: 8888
```

oxan, 2021-09-30:
Great. As I don't think there's anything actionable here, I'll close this issue now.

closed 2021-09-30 by oxan