oxan/esphome-stream-server #10: ESP32-C3 support

zuidwijk, 2021-11-21:
When trying to add stream server to an ESP32-C3, I got this error:

```yml
esphome clean esp32c3-basic.yaml
INFO Reading configuration esp32c3-basic.yaml...
Failed config

async_tcp: None
  {}
```
&&
```yml
esphome run esp32c3-basic.yaml  
INFO Reading configuration esp32c3-basic.yaml...
Failed config

async_tcp: None
  {}
```
The config I use:
```yml
esphome:
  name: esp32c3-basic
  comment: "esp32c3-basic"
  platformio_options:
    board_build.f_flash: 40000000L
    board_build.flash_mode: dio
    board_build.flash_size: 4MB

esp32:
  variant: ESP32C3
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: recommended

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

wifi:
  ssid: "********"
  password: "********"

uart:
  id: uart_tcp
  rx_pin: GPIO20
  tx_pin: GPIO21
  baud_rate: 115200
  data_bits: 8
  parity: NONE
  stop_bits: 1

external_components:
  - source: github://oxan/esphome-stream-server

stream_server:
   uart_id: uart_tcp
   port: 23
```

oxan, 2021-11-21:
Yeah, this component needs to be rewritten to use the new `socket` abstraction instead of `async_tcp`, but I haven't had the time and motivation to do that yet. PR's welcome.

oxan, 2023-02-03:
I've now (finally) finished the rewrite using the `socket` abstraction, in PR #25. That should work on the C3 now as well.

oxan, 2023-02-07:
This has now been merged into master.

closed 2023-02-07 by oxan