oxan/esphome-stream-server #11: DSMR component breaks stream server

sanderpleijers, 2021-11-26:
When i use the DSMR component of ESPhome, this serial stream server doesn't work anymore. Nothing is posted on port 8088 in this case. This is the used config:

````
external_components:
  - source: github://oxan/esphome-stream-server
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ "dsmr" ]
    refresh: 60s

*SNIP*

# Custom uart settings for DSMR v2.2
uart:

  debug:
    direction: RX
    after:
      delimiter: "\r\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
         
  id: uart_bus
  rx_pin: GPIO3
  #tx_pin: D0
  baud_rate: 9600
  data_bits: 7
  parity: Even
  stop_bits: 1

stream_server:
   uart_id: uart_bus
   port: 8088

dsmr:
  crc_check: false
```

Without the DSMR component everything works great. Can we do anything to fix this? I would like to use them together.

oxan, 2021-11-26:
You can't share a UART bus between two components, that's a fairly fundamental limitation of how ESPHome currently handles UART buses.

sanderpleijers, 2021-11-26:
Makes sense. It would be alot better if i just could grab the read out raw message from DSMR component to a text_sensor. Then i could use it from there, to http post it to my 'DSMR Reader' docker. But that is not something this component does, ill have to ask the devs at ESPhome... ;)

Thanks for the rapid reply!

closed 2021-11-26 by oxan

hmmbob, 2023-03-15:
I was running into this issue as well. Is there already a fix or feasible workaround available for this issue?

The use case is that I have a "SlimmeLezer" hooked up to my meter. I'd like this to do two things:
- Send accurate L1/L2/L3 readings to my SmartEVSE (car charger) directly from the SlimmeLezer, for which _I think I need_ the `DSMR` component in ESPHome.
- Send all P1 readings to my DSMR-Reader docker instance on a different host, to keep track of usage.

Right now, when enabling `dsmr` and `stream-server` together, I get a lot of CRC errors. Enabling just one of the two, all is fine.