oxan/esphome-stream-server #12: Build fails using latest ESPHome: src/stream_server.cpp:59:65: error: 'min' was not declared in this scope

jmartens, 2021-11-30:
Current ESPHome (v2021.11.4) fails to build your streamserver component with the following error:

```
Compiling /data/p1/.pioenvs/p1/src/stream_server.cpp.o
src/stream_server.cpp: In member function 'void StreamServerComponent::read()':
src/stream_server.cpp:59:65: error: 'min' was not declared in this scope
         size_t read = this->stream_->readBytes(buf, min(len, 128));
                                                                 ^
src/stream_server.cpp:59:65: note: suggested alternative:
In file included from /data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/algorithm:62:0,
                 from src/esphome/core/optional.h:19,
                 from src/esphome/core/component.h:7,
                 from src/stream_server.h:19,
                 from src/stream_server.cpp:17:
/data/cache/platformio/packages/toolchain-xtensa/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algo.h:4226:5: note:   'std::min'
     min(initializer_list<_Tp> __l, _Compare __comp)
     ^
In file included from src/stream_server.cpp:19:0:
src/stream_server.cpp: In member function 'virtual void StreamServerComponent::dump_config()':
src/stream_server.cpp:75:64: error: 'network_get_address' was not declared in this scope
     ESP_LOGCONFIG(TAG, "  Address: %s:%u", network_get_address().c_str(), this->port_);
                                                                ^
src/esphome/core/log.h:97:90: note: in definition of macro 'esph_log_config'
   esp_log_printf_(ESPHOME_LOG_LEVEL_CONFIG, tag, __LINE__, ESPHOME_LOG_FORMAT(format), ##__VA_ARGS__)
                                                                                          ^
src/stream_server.cpp:75:5: note: in expansion of macro 'ESP_LOGCONFIG'
     ESP_LOGCONFIG(TAG, "  Address: %s:%u", network_get_address().c_str(), this->port_);
     ^
*** [/data/p1/.pioenvs/p1/src/stream_server.cpp.o] Error 1
========================= [FAILED] Took 19.15 seconds =========================
```

oxan, 2021-11-30:
That's an old version that gets used. Try clearing the `.esphome` directory and rebuilding.

jmartens, 2021-11-30:
> That's an old version that gets used. Try clearing the `.esphome` directory and rebuilding.

Thanks for the reply, Any pointers on how to do so when using supervised?

oxan, 2021-11-30:
> Any pointers on how to do so when using supervised?

Not sure, I never used it myself.

tube0013, 2021-11-30:
directory exists at /config/esphome/.esphome on supervised install.

closed 2023-02-07 by oxan