oxan/esphome-stream-server #14: Issues with most recent version - re flashing cc2652 over network

tube0013, 2022-01-06:
So I've noticed that a newer ESPHome build using the most recent code here for the serial is not allowing successful flashes of the cc2652 over the network. I have tested this on several Olimex ESP32-PoE devices and have yet to get one to flash cleanly.

I re-flashed an older binary based on earlier code from the gist, and it completes the flash with no problem. Not sure if there is something from the update for ESPHome that is significant in the serial connection, but wanted to mention it here.

tube0013, 2022-01-07:
I was able to roll back an install of ESPHome to 2021.9.4 and compile with the current version and successfully flash over the network serial connection. So I'm thinking this is more to do with current ESPHome builds?

Will continue doing some tests.

dartanidi, 2022-01-11:
@tube0013 Looks like what is happening here:

syssi/esphome-zb-gw03/issues/8

joshuaspence, 2022-01-18:
I think I am seeing this as well.

```
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Mem Read (0x2A)'
```

tube0013, 2022-03-21:
All seems to be working more reliably with 2022.03.0. continuing to test it for a bit.

@joshuaspence I pulled in your sensor branch to try as well, It is working well.

tube0013, 2022-03-30:
after some further testing, ESPhome 2021.10 and after just is not a reliable with this component. Its better than it was, but I could not get serial to stay connected for more that 11/12 hours before a disconnect - most of the times even less. With ESPhome 2021.9.x i'm up to 36 straight hours with no disconnects.

I tried the wip sockets branch but obviously it's still a WIP. hoping this can improve things as would really like to support newer esphome features.

tube0013, 2022-07-20:
closing to open new issue re unreliable connections.

closed 2022-07-20 by tube0013