oxan/esphome-stream-server #21: Unreliable connections with ESPHome > 2021.9

tube0013, 2022-07-20:
Serial connections are very unreliable when using this component with any ESPHome Version greater than 2021.9. By unreliable I mean the connection resets every few hours.

robinsmidsrod, 2022-09-14:
Do you have any error messages or related information for when the issue happens? When the connection is reset, what do you have to do to get it working again?

tube0013, 2022-09-14:
Connection will get reestablished.

Problem being serial is disconnected frequently for small amounts of time while reconnects and zigbee network restarted.

povlhp, 2023-01-06:
Just installed esphome with this on esp32dev board. 
Inconsistent intervals. Might run 36 hours of 10 minutes. Restart will improve.

oxan, 2023-02-03:
You could try the `socket-abstraction` branch (see PR #25). I've done a major update to use the socket APIs directly instead of through the Arduino libraries, and I hope that also improves reliability. I haven't done any long duration testing myself yet though.

oxan, 2023-02-07:
I'll close this now, as I haven't been able to reproduce this with the new socket-based approach. If you still encounter this issue, please reopen the issue and we'll see if we can dive further into it.

closed 2023-02-07 by oxan