oxan/esphome-stream-server #22: Socket Version

cbr303, 2022-08-23:
Hello, any chance that the socket version will be finished? That would be very nice. Actual version doesn't start (but compile).

Christoph

cbr303, 2022-08-23:
src/esphome/components/stream_server/stream_server.cpp: In member function 'virtual void StreamServerComponent::setup()':
src/esphome/components/stream_server/stream_server.cpp:40:5: warning: missing initializer for member 'sockaddr_in::sin_zero' [-Wmissing-field-initializers]
     };
     ^

cbr303, 2022-08-23:
This happens if socket version is added:

[21:38:18]E (10935) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[21:38:18]E (10935) task_wdt:  - loopTask (CPU 1)
[21:38:18]E (10935) task_wdt: Tasks currently running:
[21:38:18]E (10935) task_wdt: CPU 0: IDLE
[21:38:18]E (10935) task_wdt: CPU 1: IDLE
[21:38:18]E (10935) task_wdt: Aborting.

oxan, 2023-02-03:
This should (finally) be fixed by bd516623ec.

oxan, 2023-02-07:
This has now been merged.

closed 2023-02-07 by oxan