oxan/esphome-stream-server #23: can use muti serial?

raxel0510, 2022-08-25:
ESP32 have 3 uart
can use
uart1  port:5000
uart2  port:5001???

RoganDawes, 2022-12-15:
It is definitely possible to run multiple stream servers, even if you do have to change the code a little to allow it. I took a copy of this code a couple of years ago, modified the constructor to accept a port parameter rather than defaulting to a set port, and simply created two stream servers. Not sure if the current code has been updated to allow the same, but it is not difficult to do yourself.

oxan, 2023-02-07:
Yes, this is possible:

```
uart:
  - id: uart1
    # ...
  - id: uart2
    # ...

stream_server:
  - uart_id: uart1
    port: 5000
  - uart_id: uart2
    port: 5001
```

closed 2023-02-07 by oxan