oxan/esphome-stream-server #30: request for telnet support

0x5e, 2023-02-20:
Thanks @oxan , the project saves my day :)
I'm using this component to remote access another linux embedded device. It would be great if telnet option things can be supported.

I've searched on the github and found this: https://github.com/yasheena/telnetspy/blob/master/TelnetSpy.cpp
I have little c++ and telnet knowledges, is it hard to migrate to this project? If anything I can help, please let me know.
Thanks again, really appreciate it.

oxan, 2023-02-20:
I don't see how the telnet protocol would be relevant here. Telnet as a protocol deals mostly with terminals, while this component just transfers data. If the device you've connected supports the telnet protocol, then you can connect using a telnet client to it; and it should just work.

0x5e, 2023-02-21:
I'm connecting esp32's gpio to another device's uart, using telnet client on the computer. it works well execpt for ctrl-c, up/down/left/right/tab these special commands. are they working when direct access uart?

oxan, 2023-03-26:
Things like ctrl-c and arrow keys are usually handled by your terminal program. The stream server just transmits them verbatim, so if the other device supports them it should already work, but it likely doesn't.

closed 2023-03-26 by oxan