oxan/esphome-stream-server #31: Official releases?

jmartens, 2023-02-20:
Could you create official releases or at least tags? That way it is easier to pin streamserver to a specific commit?

In my case it seems my local ESP device is not working with the current tip of master, but it is very hard to downgrade streamserver.

External components only tage branches or tags apparently if I consult the docs at https://esphome.io/components/external_components.html

oxan, 2023-02-20:
You can actually specify any git ref (including a commit sha) using external components, it's probably documented as branch or tag only to keep it understandable for beginners.

However, actually tagging a release isn't a bad idea. If I do that, it'll be current master though, not an old version.

> In my case it seems my local ESP device is not working with the current tip of master

What's not working?

jmartens, 2023-02-20:
Somehow I lost all transmission of P1 smart energy meter after january 28th at least no data is present anymore in the Energy panel.

Still debugging why, so far I can see multiple possible causes:
- changes to the serialserver code base since the last working update of my ESP device, hence my request for releases. Will see if working with a commit hash makes any difference, at least I can perhaps pinpoint if it is the code base, at what point in the code base it might be introduced.
- no telegrams are actually transmitted to the ESP device anymore
- ESPHome encryption feature needs implementation
- ... perhaps something else I overlooked

jmartens, 2023-02-20:
> You can actually specify any git ref (including a commit sha) for external components, it's probably documented as branch or tag to keep it understandable for beginners.

Have not seem to get it working, either:
```
  Remote branch master@43ade65d088d99a6f1bb2da32f7cdd666eb4eb59 not found in upstream origin.
  source: 
    type: git
    url: https://github.com/oxan/esphome-stream-server
    ref: master@43ade65d088d99a6f1bb2da32f7cdd666eb4eb59
```

or 
```
  Remote branch 43ade65d088d99a6f1bb2da32f7cdd666eb4eb59 not found in upstream origin.
  source: 
    type: git
    url: https://github.com/oxan/esphome-stream-server
    ref: 43ade65d088d99a6f1bb2da32f7cdd666eb4eb59
```
when validating/building the yaml.

jmartens, 2023-02-20:
Checked the source (https://github.com/esphome/esphome/blob/dev/esphome/git.py#L64-L75), and AFAICT they do not support specific branch/commit references

oxan, 2023-02-20:
Hmm, you're right, that's a bit disappointing it only does it for PRs.

I'll make a release when I've a bit more time to do it properly, but in the meantime I've pushed the [async-tcp](https://github.com/oxan/esphome-stream-server/tree/async-tcp) branch with the last commit pre-socket refactor which you can use for testing.

jmartens, 2023-02-28:
> * ... perhaps something else I overlooked

Indeed. Turned out to be faulty hardware, had to change a FET due to a short circuit.

joneshf, 2025-01-19:
> Checked the source (https://github.com/esphome/esphome/blob/dev/esphome/git.py#L64-L75), and AFAICT they do not support specific branch/commit references

Looks like this was changed in 2024.4.0 with this PR: https://github.com/esphome/esphome/pull/6446. So we can use commit SHAs now!

jmartens, 2025-01-19:
Nice, but I would still prefer tags and releases. Just as any other software does.