oxan/esphome-stream-server #37: Can not connect esphome over ethernet after changes from 22 march 2023

teklabania, 2023-03-24:
Hi, I'm using zigbee coordinator with ESP32 and Ethernet, like [this one](https://forum.arturhome.pl/uploads/default/original/2X/c/cdc821753880c76e0354b70bb1500729793dcf5c.jpeg). Lately I upgraded esphome on the device, and esphome-stream-server was also upgraded. The changes from your commits from 22.03.2023 made me unable to connect to my esphome device. I could only ping it but couldn't connect to web_server on port 80 nor api on port 6053
```
WARNING Can't connect to ESPHome API for device-name.my.domain: Error connecting to ('192.168.1.10', 6053): [Errno 111] Connect call failed ('192.168.1.10', 6053)
```
This is my configuration:
```
substitutions:
  device_name: cc2652p01
  friendly_name: ${device_name} over LAN

esphome:
  name: ${device_name}
  comment: ${friendly_name}
  platform: ESP32
  board: esp-wrover-kit

web_server:
  port: 80

ethernet:
  type: LAN8720
  mdc_pin: GPIO23
  mdio_pin: GPIO18
  clk_mode: GPIO0_IN
  phy_addr: 1
  power_pin: GPIO16

external_components:
  - source: github://oxan/esphome-stream-server

stream_server:
  uart_id: uart_bus

uart:
  id: uart_bus
  rx_pin: GPIO5
  tx_pin: GPIO17
  baud_rate: 115200
```
I forked your repository and created a tag at a commit `53a7260e0158f99f57763a7e41eb76afac309506` from 07.02.2023, then set up `external_components.source` pointing to my repository with reference to the tag I created and everything works perfectly.

piotrrekkk, 2023-03-24:
Confirmed, I had same issue.

oxan, 2023-03-24:
Crap, I messed up. Will fix tomorrow. You can define both sensors for now to workaround it.

teklabania, 2023-03-24:
@oxan , how about tagging your commits in the future so we could pin to the tag?

closed 2023-03-25 by oxan

skipfire, 2023-03-25:
@oxan I tried the update and it is still causing loading errors.  API and web server do not load, and the WiFi keeps bouncing on and off.

kluszczyn, 2023-03-25:
> @oxan I tried the update and it is still causing loading errors. API and web server do not load, and the WiFi keeps bouncing on and off.

same issue here

oxan, 2023-03-25:
@teklabania Yeah, that's not a bad idea; but I've had vanishingly little time to actually make a proper release, and I don't really want to just tag any random commit (especially if it has known issues). Ideally ESPHome would also allow you to pin any commit instead of just branches or tags.

@skipfire @kluszczyn Are you sure ESPHome actually downloaded the new version? by default it only checks git repositories for new versions once a week. See [documentation](https://esphome.io/components/external_components.html#refresh).

skipfire, 2023-03-25:
Yes I'm sure.  I have a test unit that I removed the stream server from, then added it back after you indicated it was fixed.  When I remove stream-server it comes back up but with the wifi instability I have to push it manually over USB if it has the current stream-server code on it.

oxan, 2023-03-25:
> I have a test unit that I removed the stream server from, then added it back after you indicated it was fixed.

That does not actually ensure ESPHome updates the stream server code with the latest version from GitHub. Can you try adding `refresh: 0s` to the external component, as explained in the docs I linked?

skipfire, 2023-03-25:
I lost the refresh 0 along the way, it is working with the new code.

kluszczyn, 2023-03-25:
I've purged `.build` folder inside esphome, recompiled, and it is working now too! Thanks!

commit e37d924e (2023-03-25): Fix null pointer dereference when no sensors are defined (fixes #37)