oxan/esphome-stream-server #39: Not listening on ESPHome 2023.4

joshuaspence, 2023-04-20:
Since ESPHome 2023.4 the `stream_server` component doesn't listen on the specified port. I think this was broken by esphome/esphome#4574.

oxan, 2023-04-20:
Yeah, that's an unfortunate change. Thanks for your PR, I've merged the fix!

closed 2023-04-20 by oxan

bugale, 2023-06-17:
@oxan @joshuaspence 
I am not really sure what happened here - the PR that was "blamed for the issue" has been only merge to `dev` branch, while it's "fix" has been merged to `master` branch.
Anyway, in current `master`, seems like there's a missing `htons` call.
This configuration:
```
uart:
  id: uart_0
  # ...

stream_server:
  uart_id: uart_0
  port: 1234
```
listens on its inverse order port:
```
>>> s.connect(('192.168.1.15', 1234))
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it
>>> hex(1234)
'0x4d2'
>>> s.connect(('192.168.1.15', 0xd204))
>>>
```

oxan, 2023-06-17:
> the PR that was "blamed for the issue" has been only merge to `dev` branch,

Of the ESPHome repository, which does not have a master branch; in any case this commit was included in the 2023.4 release.

> while it's "fix" has been merged to `master` branch.

Of the esphome-stream-server repository.

>  Anyway, in current master, seems like there's a missing htons call.

Which ESPHome version are you observing this with?

bugale, 2023-06-17:
> Of the ESPHome repository, which does not have a master branch; in any case this commit was included in the 2023.4 release.
Oh I see.

> Which ESPHome version are you observing this with?
`v2023.5.5`