oxan/esphome-stream-server #49: 2nd binary sensor for 2nd uart/stream_server

tube0013, 2024-05-31:
I'm not clear if this is possible, and if it is how to configure 2 binary sensors when using 2 serial streams. Or even to use 1 when having 2 servers:

[image]

MattWestb, 2025-07-26:
Im using this on ESP32-C3 and looks working:
```
binary_sensor:
  - platform: status
    name: "$hostname Satus"
  - platform: stream_server
    stream_server: SS1
    name: $hostname Stream Server 1 Status
  - platform: stream_server
    stream_server: SS2
    name: $hostname Stream Server 2 Status
```
with the servers defined as:
```
stream_server:
  - id: SS1
    uart_id: uart_bus1
    port: 6638
  - id: SS2
    uart_id: uart_bus2
    port: 6639
```
The only bad thing is its looks the ESPHome is getting blocked if connecting 2 strean servers and one loging to it from WEB UI or ESPHome panel and cant connect the stream server / log component (the first  2 looks working but  the third is failing of more then 2 connections to the device (API looks working OK).

MattWestb, 2025-07-26:
Great and thanks i shall trying it !!
In the earlier versions it was possible allowing more stream connections with:
```
  multi_client: true # optional, default is false. Set to true to allow multiple simultaneous connections
´´´
but its taken away but it was great then having more terminals for sending / receiving commands and at the same time looking on logs (like hacking the LIDL ZbGw).

MattWestb, 2025-07-26:
Working set:
```
esp32:
  board: esp32-c3-devkitm-1
  framework:
#    type: arduino
    type: esp-idf
    sdkconfig_options:
      CONFIG_LWIP_MAX_SOCKETS: "16"
```
Now i can have 2 stream servers connected and more then one terminal loging (ESPHome dashboard and web page).

Great thanks now i can converting some ESP8266 D1 Mini / ESP-01 ESP8285 devices to ESP32-C/S3 modules.