oxan/esphome-stream-server #52: TCP receive stops after  ~256Kb

bJ3h2-4-8, 2024-06-27:
I have a reproducible situation with esphome (sha256:6e659021515743b3d0d988084edcf31a777829f23e2d1c13b87901fa899bbce6) where the Socket of esphome stops receiving data. I'm not sure if this a esphome, or a socket setup issue. After reconnecting I was able to continue to send the next 256kb where it again stop receiving.

jdwhite, 2024-06-27:
I have the same issue. I don't know precisely how much data is received before it stops and as I'm currently on vacation I probably won't be able to debug this further until after July 4th.

xNUTx, 2024-07-16:
Same issue here, it will stop working after a while. It started happening when I updated to 2024.6.1, but it was still running 2024.5.x when I updated it...

jdwhite, 2024-10-28:
Using ESPhome 2024.10.2 I no longer experience this issue. Anyone else?