oxan/esphome-stream-server #54: Component uart took a long time for an operation

metaljay, 2024-09-13:
Hi,

My config flashes perfectly and ive enabled debug, here is my code:

```javascript
esphome:
  name: texecom-link
  friendly_name: texecom-link

esp8266:
  board: d1

external_components:
  - source: github://oxan/esphome-stream-server

stream_server:
  - uart_id: uart_bus
    port: 10001
    id: "b1"

binary_sensor:
  - platform: stream_server
    stream_server: "b1"
    connected:
      name: "serial_server_1"
      
uart:
  - id: uart_bus
    debug:
      direction: BOTH
    tx_pin: GPIO1
    rx_pin: GPIO3
    baud_rate: 19200 
    data_bits: 8
    stop_bits: 2

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  password: ""

ota:
  - platform: esphome
    password: ""

wifi:
  ssid: "XXX"
  password: "XXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Texecom Fallback Hotspot"
    password: "4MycrSSwOufn"

captive_portal:
    
``` 

I am trying to connect via texecom2MQTT which fails to connect with the following error
```javascript
2024-09-13 13:54:14 - INFO: Starting texecom2mqtt v1.2.3 (Node v16.13.0)...
2024-09-13 13:54:14 - INFO: Connected to alarm, sleeping for 2 seconds...
2024-09-13 13:54:16 - DEBUG: Executing serial number command
2024-09-13 13:54:20 - DEBUG: Command serial timed out (attempt 1, id: serial).
2024-09-13 13:54:23 - DEBUG: Command serial timed out (attempt 2, id: serial).
2024-09-13 13:54:27 - DEBUG: Command serial timed out (attempt 3, id: serial).
2024-09-13 13:54:30 - DEBUG: Command serial timed out (attempt 4, id: serial).
2024-09-13 13:54:34 - DEBUG: Command serial timed out (attempt 5, id: serial).
2024-09-13 13:54:34 - ERROR: Unhandled rejection - Command serial timed out 5 times and could not be completed
``` 

When I look at the log for ESPhome-stream-server I see this: (192.168.2.254 is my Homeassistant raspberry pi and .1.29 is my macbook)
```javascript
[14:54:14][D][stream_server:081]: New client connected from 192.168.2.254
[14:54:16][D][uart_debug:114]: >>> 03:5A:A2
[14:54:20][D][uart_debug:114]: >>> 03:5A:A2
[14:54:23][D][uart_debug:114]: >>> 03:5A:A2
[14:54:27][D][uart_debug:114]: >>> 03:5A:A2
[14:54:30][D][uart_debug:114]: >>> 03:5A:A2
[14:54:34][D][stream_server:163]: Client 192.168.2.254 disconnected
[14:55:12][D][stream_server:163]: Client 192.168.1.29 disconnected
[14:55:12][D][binary_sensor:036]: 'serial_server_1': Sending state OFF
[14:55:12][D][stream_server:081]: New client connected from 192.168.1.29
[14:55:12][D][binary_sensor:036]: 'serial_server_1': Sending state ON
[14:55:12][D][stream_server:081]: New client connected from 192.168.1.29
[14:55:13][D][uart_debug:114]: >>> 47:45:54:20:2F:20:48:54:54:50:2F:31:2E:31:0D:0A:48:6F:73:74:3A:20:74:65:78:65:63:6F:6D:2D:6C:69:6E:6B:2E:6C:6F:63:61:6C:3A:31:30:30:30:31:0D:0A:55:70:67:72:61:64:65:2D:49:6E:73:65:63:75:72:65:2D:52:65:71:75:65:73:74:73:3A:20:31:0D:0A:41:63:63:65:70:74:3A:20:74:65:78:74:2F:68:74:6D:6C:2C:61:70:70:6C:69:63:61:74:69:6F:6E:2F:78:68:74:6D:6C:2B:78:6D:6C:2C:61:70:70:6C:69:63:61:74:69:6F:6E:2F:78:6D:6C:3B:71:3D:30:2E:39:2C:2A:2F:2A:3B:71:3D:30:2E:38:0D
[14:55:13][D][uart_debug:114]: >>> 0A:55:73:65:72:2D:41:67:65:6E:74:3A:20:4D:6F:7A:69:6C:6C:61:2F:35:2E:30:20:28:4D:61:63:69:6E:74:6F:73:68:3B:20:49:6E:74:65:6C:20:4D:61:63:20:4F:53:20:58:20:31:30:5F:31:35:5F:37:29:20:41:70:70:6C:65:57:65:62:4B:69:74:2F:36:30:35:2E:31:2E:31:35:20:28:4B:48:54:4D:4C:2C:20:6C:69:6B:65:20:47:65:63:6B:6F:29:20:56:65:72:73:69:6F:6E:2F:31:37:2E:36:20:53:61:66:61:72:69:2F:36:30:35:2E:31:2E:31:35:0D:0A:41:63:63:65:70:74:2D:4C:61:6E:67:75:61:67:65:3A:20:65
[14:55:13][W][component:237]: Component stream_server took a long time for an operation (825 ms).
[14:55:13][W][component:238]: Components should block for at most 30 ms.
[14:55:13][D][uart_debug:114]: >>> 6E:2D:47:42:2C:65:6E:3B:71:3D:30:2E:39:0D:0A:41:63:63:65:70:74:2D:45:6E:63:6F:64:69:6E:67:3A:20:67:7A:69:70:2C:20:64:65:66:6C:61:74:65:0D:0A:43:6F:6E:6E:65:63:74:69:6F:6E:3A:20:6B:65:65:70:2D:61:6C:69:76:65:0D:0A:0D:0A
[14:55:13][W][component:237]: Component uart took a long time for an operation (141 ms).
[14:55:13][W][component:238]: Components should block for at most 30 ms.
[14:55:36][D][stream_server:163]: Client 192.168.1.29 disconnected
[14:56:00][D][stream_server:163]: Client 192.168.1.29 disconnected
[14:56:00][D][binary_sensor:036]: 'serial_server_1': Sending state OFF

``` 

Any ideas where I am going wrong?

jdwhite, 2024-09-13:
In my experience, the "component took a long time" warnings can be ignored. I've seen them on other integrations and, at least in those, they were harmless. It's good to know these things when first setting up, but it's generally (again, IMO) not very actionable and I've taken to suppressing all component messages in my config.

logger:
  level: DEBUG
  logs:
    econet: NONE    # Zorch packet dumps.
    component: NONE # Zorch "took too long" messages.

Yes, this doesn't just suppress the "took too long" warnings, but everything from the component. My comment is to remind me why I set it to NONE.

Ideally, I'd just suppress Warnings from component, but I haven't figured out how or if it's possible yet.

-Jason

> 
> 
> Hi,
> 
> My config flashes perfectly and ive enabled debug, here is my code:
> 
> esphome:
>   name: texecom-link
>   friendly_name: texecom-link
> 
> esp8266:
>   board: d1
> 
> external_components:
>   - source: github://oxan/esphome-stream-server
> 
> stream_server:
>   - uart_id: uart_bus
>     port: 10001
>     id: "b1"
> 
> binary_sensor:
>   - platform: stream_server
>     stream_server: "b1"
>     connected:
>       name: "serial_server_1"
>       
> uart:
>   - id: uart_bus
>     debug:
>       direction: BOTH
>     tx_pin: GPIO1
>     rx_pin: GPIO3
>     baud_rate: 19200 
>     data_bits: 8
>     stop_bits: 2
> 
> # Enable logging
> logger:
>   level: DEBUG
> 
> # Enable Home Assistant API
> api:
>   password: ""
> 
> ota:
>   - platform: esphome
>     password: ""
> 
> wifi:
>   ssid: "XXX"
>   password: "XXX"
> 
>   # Enable fallback hotspot (captive portal) in case wifi connection fails
>   ap:
>     ssid: "Texecom Fallback Hotspot"
>     password: "4MycrSSwOufn"
> 
> captive_portal:
>     
> I am trying to connect via texecom2MQTT which fails to connect with the following error
> 
> 2024-09-13 13:54:14 - INFO: Starting texecom2mqtt v1.2.3 (Node v16.13.0)...
> 2024-09-13 13:54:14 - INFO: Connected to alarm, sleeping for 2 seconds...
> 2024-09-13 13:54:16 - DEBUG: Executing serial number command
> 2024-09-13 13:54:20 - DEBUG: Command serial timed out (attempt 1, id: serial).
> 2024-09-13 13:54:23 - DEBUG: Command serial timed out (attempt 2, id: serial).
> 2024-09-13 13:54:27 - DEBUG: Command serial timed out (attempt 3, id: serial).
> 2024-09-13 13:54:30 - DEBUG: Command serial timed out (attempt 4, id: serial).
> 2024-09-13 13:54:34 - DEBUG: Command serial timed out (attempt 5, id: serial).
> 2024-09-13 13:54:34 - ERROR: Unhandled rejection - Command serial timed out 5 times and could not be completed
> When I look at the log for ESPhome-stream-server I see this: (192.168.2.254 is my Homeassistant raspberry pi and .1.29 is my macbook)
> 
> [14:54:14][D][stream_server:081]: New client connected from 192.168.2.254
> [14:54:16][D][uart_debug:114]: >>> 03:5A:A2
> [14:54:20][D][uart_debug:114]: >>> 03:5A:A2
> [14:54:23][D][uart_debug:114]: >>> 03:5A:A2
> [14:54:27][D][uart_debug:114]: >>> 03:5A:A2
> [14:54:30][D][uart_debug:114]: >>> 03:5A:A2
> [14:54:34][D][stream_server:163]: Client 192.168.2.254 disconnected
> [14:55:12][D][stream_server:163]: Client 192.168.1.29 disconnected
> [14:55:12][D][binary_sensor:036]: 'serial_server_1': Sending state OFF
> [14:55:12][D][stream_server:081]: New client connected from 192.168.1.29
> [14:55:12][D][binary_sensor:036]: 'serial_server_1': Sending state ON
> [14:55:12][D][stream_server:081]: New client connected from 192.168.1.29
> [14:55:13][D][uart_debug:114]: >>> 47:45:54:20:2F:20:48:54:54:50:2F:31:2E:31:0D:0A:48:6F:73:74:3A:20:74:65:78:65:63:6F:6D:2D:6C:69:6E:6B:2E:6C:6F:63:61:6C:3A:31:30:30:30:31:0D:0A:55:70:67:72:61:64:65:2D:49:6E:73:65:63:75:72:65:2D:52:65:71:75:65:73:74:73:3A:20:31:0D:0A:41:63:63:65:70:74:3A:20:74:65:78:74:2F:68:74:6D:6C:2C:61:70:70:6C:69:63:61:74:69:6F:6E:2F:78:68:74:6D:6C:2B:78:6D:6C:2C:61:70:70:6C:69:63:61:74:69:6F:6E:2F:78:6D:6C:3B:71:3D:30:2E:39:2C:2A:2F:2A:3B:71:3D:30:2E:38:0D
> [14:55:13][D][uart_debug:114]: >>> 0A:55:73:65:72:2D:41:67:65:6E:74:3A:20:4D:6F:7A:69:6C:6C:61:2F:35:2E:30:20:28:4D:61:63:69:6E:74:6F:73:68:3B:20:49:6E:74:65:6C:20:4D:61:63:20:4F:53:20:58:20:31:30:5F:31:35:5F:37:29:20:41:70:70:6C:65:57:65:62:4B:69:74:2F:36:30:35:2E:31:2E:31:35:20:28:4B:48:54:4D:4C:2C:20:6C:69:6B:65:20:47:65:63:6B:6F:29:20:56:65:72:73:69:6F:6E:2F:31:37:2E:36:20:53:61:66:61:72:69:2F:36:30:35:2E:31:2E:31:35:0D:0A:41:63:63:65:70:74:2D:4C:61:6E:67:75:61:67:65:3A:20:65
> [14:55:13][W][component:237]: Component stream_server took a long time for an operation (825 ms).
> [14:55:13][W][component:238]: Components should block for at most 30 ms.
> [14:55:13][D][uart_debug:114]: >>> 6E:2D:47:42:2C:65:6E:3B:71:3D:30:2E:39:0D:0A:41:63:63:65:70:74:2D:45:6E:63:6F:64:69:6E:67:3A:20:67:7A:69:70:2C:20:64:65:66:6C:61:74:65:0D:0A:43:6F:6E:6E:65:63:74:69:6F:6E:3A:20:6B:65:65:70:2D:61:6C:69:76:65:0D:0A:0D:0A
> [14:55:13][W][component:237]: Component uart took a long time for an operation (141 ms).
> [14:55:13][W][component:238]: Components should block for at most 30 ms.
> [14:55:36][D][stream_server:163]: Client 192.168.1.29 disconnected
> [14:56:00][D][stream_server:163]: Client 192.168.1.29 disconnected
> [14:56:00][D][binary_sensor:036]: 'serial_server_1': Sending state OFF
> Any ideas where I am going wrong?

metaljay, 2024-09-13:
Will give that a try but I don’t think that will identify what’s causing the connection issue? (Sorry not trying to sound stupid)

jdwhite, 2024-09-13:
> Will give that a try but I don’t think that will identify what’s causing the connection issue? (Sorry not trying to sound stupid)

Yes, I did latch on to the warning message in your logs, but it was in line with the subject of your issue which wasn't anything to do with not being able to connect. My bad.

Your clients are successfully connecting to ESPHome, but the `uart_debug` log entries indicate that ESPhome is sending data over the UART (`>>>`), but never receiving a response (`<<<`). This likely indicates a misconfiguration of the UART parameters (double check `tx_pin`/`rx_pin`), wiring the TX/RX pins backward or to the wrong pins, or a combination thereof.

Try swapping the TX and RX wires. This may also be useful in checking your pin names: https://esphome.io/guides/configuration-types#config-pin

jdwhite, 2024-09-14:
I believe I know what's wrong. I think the `- id:` in your UART config, while valid YAML, is not valid syntax for ESPHome. I re-read the docs on the [UART Bus - ESPHome](https://esphome.io/components/uart.html) and looked at all of my ESPHome instances that use a UART. Two of them don't even use the `id` variable as it's optional for a single uart.

Try changing `- id:` to just `id:` and indent everything under `uart:` by two spaces.

Here's my working serial streamer UART config block:
```
uart:
  id: uart_bus_1
  # Logger enabled, so don't use UART1.
  #tx_pin: 1
  #rx_pin: 3
  # Use UART2.
  tx_pin: 17
  rx_pin: 16
  baud_rate: 115200
```

metaljay, 2024-09-15:
I think the issue is actually the hardware, not the software. ill close this as genuinely think software is fine.

closed 2024-09-15 by metaljay